xshellinc/iotit #4: Cannot connect to edison

makeupsomething, 2017-02-17:
When prompted to connect and reboot device `itoit` does not connect to edison

```
[+] Would you like to flash your device? (y/yes OR n/no): yes
The authenticity of host '[localhost]:2222 ([127.0.0.1]:2222)' can't be established.
ECDSA key fingerprint is SHA256:tfjGcdYwG7a9rW0xWcoUmazClOpOfNerZAJwQJOa1VY.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '[localhost]:2222' (ECDSA) to the list of known hosts.
Using U-Boot target: edison-blankcdc
Now waiting for dfu device 8087:0a99
Please plug and reboot the board
Timed out while waiting for dfu device 8087:0a99
DEBUG: lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
DEBUG: dfu-util -l
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Did you plug and reboot your board?
If yes, please try a recovery by calling this script with the --recovery option
[+] Stopping virtual machine - Name:isaax-box-edison UUID:8941a5a1-d731-4981-a6fc-8a8fc1ef2205
NOTE: You might need to run `sudo ifconfig {interface} 192.168.2.2` in order to access Edison at 192.168.2.15
```

[iotit.log.zip](https://github.com/xshellinc/iotit/files/782807/iotit.log.zip)

larenso, 2017-02-17:
@makeupsomething it happens from occasion to occasion correct?

makeupsomething, 2017-02-17:
@larenso Every time, so maybe its something that I'm doing wrong.

larenso, 2017-02-22:
Bad usb cable could cause problems detecting Edison and as a consequence flashing/configuring it

makeupsomething, 2017-02-22:
Yep, and need to be care about interface settings. Please add to the readme 'Common Problems' section.

closed 2017-02-22 by makeupsomething