xshellinc/iotit #6: Adding default DNS server

kylycht, 2017-02-24:
There was a problem installing dependencies on a device with our `script` service
The issue comes with connecting to the repo

Proposed solution is to add a `8.8.8.8` as default DNS server during `device init`

I am opening this as a discussion since some might think it is a user's responsibility to configure it

Here is issue that was encountered:
`{"level":"debug","msg":"Publishing ({\"payload\":{\"name\":\"pre_install\",\"stdout\":\"I run before user application is installed\\nupdate\\nGet:1 http://archive.raspberrypi.org jessie InRelease [22.9 kB]\\nGet:2 http://archive.raspberrypi.org jessie/main armhf Packages [144 kB]\\nGet:3 http://archive.raspberrypi.org jessie/ui armhf Packages [54.8 kB]\\nIgn http://archive.raspberrypi.org jessie/main Translation-en\\nIgn http://archive.raspberrypi.org jessie/ui Translation-en\\nGet:4 http://mirrordirector.raspbian.org jessie InRelease [14.9 kB]\\nErr http://mirrordirector.raspbian.org jessie/main armhf Packages\\n  Unable to connect to mirrordirector.raspbian.org:http:\\nErr http://mirrordirector.raspbian.org jessie/contrib armhf Packages\\n  Unable to connect to mirrordirector.raspbian.org:http:\\nErr http://mirrordirector.raspbian.org jessie/non-free armhf Packages\\n  Unable to connect to mirrordirector.raspbian.org:http:\\nErr http://mirrordirector.raspbian.org jessie/rpi armhf Packages\\n  Unable to connect to mirrordirector.raspbian.org:http:\\nErr http://mirrordirector.raspbian.org jessie/contrib Translation-en\\n  Unable to connect to mirrordirector.raspbian.org:http:\\nErr http://mirrordirector.raspbian.org jessie/main Translation-en\\n  Unable to connect to mirrordirector.raspbian.org:http:\\nErr http://mirrordirector.raspbian.org jessie/non-free Translation-en\\n  Unable to connect to mirrordirector.raspbian.org:http:\\nErr http://mirrordirector.raspbian.org jessie/rpi Translation-en\\n  Unable to connect to mirrordirector.raspbian.org:http:\\nFetched 237 kB in 4min 1s (981 B/s)\\n\",\"stderr\":\"W: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/main/binary-armhf/Packages  Unable to connect to mirrordirector.raspbian.org:http:\\n\\nW: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/contrib/binary-armhf/Packages  Unable to connect to mirrordirector.raspbian.org:http:\\n\\nW: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/non-free/binary-armhf/Packages  Unable to connect to mirrordirector.raspbian.org:http:\\n\\nW: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/rpi/binary-armhf/Packages  Unable to connect to mirrordirector.raspbian.org:http:\\n\\nW: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/contrib/i18n/Translation-en  Unable to connect to mirrordirector.raspbian.org:http:\\n\\nW: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/main/i18n/Translation-en  Unable to connect to mirrordirector.raspbian.org:http:\\n\\nW: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/non-free/i18n/Translation-en  Unable to connect to mirrordirector.raspbian.org:http:\\n\\nW: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/jessie/rpi/i18n/Translation-en  Unable to connect to mirrordirector.raspbian.org:http:\\n\\nE: Some index files failed to download. They have been ignored, or old ones used instead.\\n\",\"exitCode\":100},\"type\":\"scriptCompleted\"})","time":"2017-02-24T07:13:12Z"}`

bnuriddin, 2017-02-24:
someone thinks that DNS issue should be resolved on DNS server itself, not on the board itself.

Device might be running on the network with bad connectivity to Goggle DNS or user running devices in private network and we putting user to the situation where he got no other choice than allowing Goggle DNS, which is not good

makeupsomething, 2017-02-24:
Moving this issue to issax main

kylycht, 2017-02-24:
And your proposed solution is to leave it up to user to resolve DNS issue? 

If you disagree, please add your alternative solution

makeupsomething, 2017-02-24:
Issue moved to [xshellinc/isaax-main #119](https://github.com/xshellinc/isaax-main/issues/119) via [**ZenHub**](https://www.zenhub.com/)

larenso, 2017-03-10:
@dbsparkle @kylycht 
seems that configs are written only if you configure a board, if you skip this step `would you like to configure your board (y/n): n` then the secondary dns is not configured

is that ok?

dbsparkle, 2017-03-10:
yeah, but lets add option in `would you like to configure your board (y/n):y`

Something like "Would you like to set a secondary DNS? (y/n):" 

And if answer is `yes` then `Enter an IP address for a secondary DNS`

https://github.com/xshellinc/iotit/issues/6#issuecomment-285578586

larenso, 2017-03-10:
@dbsparkle I have added 8.8.8.8 as a secondary dns silently, so no need to enter ip address as explained in the description, or should I ask `would you like to setup 8.8.8.8 as a secondary dns?`

dbsparkle, 2017-03-10:
Yeah, user must be aware of that kind of stuff.

https://github.com/xshellinc/iotit/issues/6#issuecomment-285592488

closed 2017-03-21 by larenso