xshellinc/iotit #7: Rename 'isaax' to 'iotit'

dbsparkle, 2017-02-25:
```
[?] Select device type: 1
[+] flashing raspberry-pi
[+] Checking virtual machine
[+] Starting virtual machine download
[+] Download isaax-box-sd.zip 141.33 MB / 142.46 MB   99.21% 0s                                     
/Users/domi/.isaax/virtualbox/sd/0.1.1/isaax-box-sd.zip https://s3-ap-northeast-1.amazonaws.com/isaax-distro/vm/sd/0.1.1/isaax-box-sd.zip
[+] Download isaax-box-sd.zip 142.46 MB / 142.46 MB  100.00% 0s                                     
[+] Unzipping /Users/domi/.isaax/virtualbox/sd/0.1.1/isaax-box-sd.zip to /Users/domi/VirtualBox VMs
[+] Unzipping isaax-box-sd.zip 403.01 MB / 403.01 MB  100.00% 6s                
[+] Done
[+] Registering isaax-box-sd
[+] Done
[+] No problem!
[+] Checking running virtual machine
```

larenso, 2017-02-28:
@bnuriddin could you pls rename packages on the s3

larenso, 2017-02-28:
@dbsparkle noticed
https://s3-ap-northeast-1.amazonaws.com/`isaax-distro`/...
should the `isaax-distro` also be changed?

dbsparkle, 2017-02-28:
Nope. Isaax-distro is a proprietary thing. It can stay as it is now. Later we may think about adding other providers as well (giving user an option where he wants to download from)

dbsparkle, 2017-03-02:
@larenso rename `.isaax` into `.iotit` 
http://prntscr.com/eeyvwv

closed 2017-03-07 by larenso