xshellinc/iotit #9: Linting is "not perfect"

dbsparkle, 2017-02-28:
Exported `const` or `var` should have comment (or unexported) :
1. https://github.com/xshellinc/iotit/blob/develop/iotit.go#L16
2. https://github.com/xshellinc/iotit/blob/develop/iotit.go#L32
3. https://github.com/xshellinc/iotit/blob/develop/iotit.go#L33

Struct field `Dns` should be `DNS` : 
1. https://github.com/xshellinc/iotit/blob/develop/device/device.go#L46

Shouldn't `type bool`  be omitted  from declaration, since it will be inferred from the right side anyway?
1. https://github.com/xshellinc/iotit/blob/develop/device/device.go#L191
2. https://github.com/xshellinc/iotit/blob/develop/device/device.go#L241

Shouldn't we drop `0` from declaration of `var num`? it is the zero value...
1. https://github.com/xshellinc/iotit/blob/develop/device/device.go#L318
2. https://github.com/xshellinc/iotit/blob/develop/device/device.go#L351
3. https://github.com/xshellinc/iotit/blob/develop/device/device.go#L352

`setIp` should be renamed into `setIP`
1. https://github.com/xshellinc/iotit/blob/develop/device/util.go#L131

Error should be the last type when returning multiple items:
1. https://github.com/xshellinc/iotit/blob/develop/device/workstation/w_linux.go#L122

`Vm...` should be `VM...`
1. https://github.com/xshellinc/iotit/blob/develop/lib/repo/repository.go#L67
2. https://github.com/xshellinc/iotit/blob/develop/lib/repo/repository.go#L71
3. https://github.com/xshellinc/iotit/blob/develop/lib/repo/repository.go#L75

larenso, 2017-02-28:
> Shouldn't type bool be omitted from declaration, since it will be inferred from the right side anyway?
Shouldn't we drop 0 from declaration of var num? it is the zero value...

this is in the progress, I am changing it to dialogs function, too many spagetti code

> Error should be the last type when returning multiple items:
setIp should be renamed into setIP
Struct field Dns should be DNS :
Vm... should be VM...

agree

larenso, 2017-03-06:
Fixed many spagetti code mostly connected with dialogs and exec commands

closed 2017-03-07 by dbsparkle