xshellinc/iotit #17: Cannot register virtual box

makeupsomething, 2017-03-06:
iotit fails to register virtual box every time when flashing device. Same for all devices.

```
daryl-no-MacBook-Pro:iotit daryl$ ./iotit
   [1] raspberry-pi
   [2] edison
   [3] nano-pi
   [4] beaglebone
[?] Select device type: 3
[+] flashing nano-pi
[+] Checking virtual machine
[+] Unzipping /Users/daryl/.iotit/virtualbox/sd/0.1.1/iotit-box-sd.zip to /Users/daryl/VirtualBox VMs
[+] Unzipping iotit-box-sd.zip 403.01 MB / 403.01 MB  100.00% 8s                
[+] Done
[+] Registering iotit-box-sd
[-] Error:  exit status 1
[-] Exiting ... 
```

larenso, 2017-03-06:
found, problem in VM was not fully renamed

larenso, 2017-03-06:
https://github.com/xshellinc/isaax-main/issues/135

makeupsomething, 2017-03-07:
working now

Note, the new VM UUID conflicts if you still have the old issax VM registered. Either we need to add a note in the readme to alert users this can happen or we version our VM's as @bnuriddin suggested.

closed 2017-03-07 by makeupsomething