xshellinc/iotit #21: Static ip not sticking on edison

makeupsomething, 2017-03-08:
When I set static ip for edison the device takes a different ip.

```
[+] Current values are:
 	[+] Address:192.168.0.254
	 [+] Network:192.168.0.0
	 [+] Gateway:192.168.0.1
	[+] Netmask:255.255.255.0
	[+] DNS:192.168.0.1
[?] Change values? (y/yes OR n/no):y
[?] IP address of the device: 192.168.10.174
[+] Validating IP address: 192.168.10.174
[+] validating: \ 
[?] Please enter your network: 192.168.10.0
[+] Validating IP address: 192.168.10.0
[?] Please enter your gateway: 192.168.10.1
[+] Validating IP address: 192.168.10.1
[?] Please enter your netmask: 255.255.255.0
[+] Validating IP address: 255.255.255.0
[?] Please enter your dns server: 192.168.10.1
[+] Validating IP address: 192.168.10.1
```

`ifconfig on edison`

```
daryl-no-MacBook-Pro:iotit daryl$ ssh root@192.168.2.15
root@192.168.2.15's password: 
root@edison:~# 
root@edison:~# 
root@edison:~# 
root@edison:~# ifconfig
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1612 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1612 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:122608 (119.7 KiB)  TX bytes:122608 (119.7 KiB)

usb0      Link encap:Ethernet  HWaddr 02:00:86:01:c6:f2  
          inet addr:192.168.2.15  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::86ff:fe01:c6f2/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:467 errors:0 dropped:0 overruns:0 frame:0
          TX packets:412 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:57641 (56.2 KiB)  TX bytes:73477 (71.7 KiB)

wlan0     Link encap:Ethernet  HWaddr 78:4b:87:a3:90:11  
          inet addr:192.168.10.145  Bcast:192.168.10.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:150 errors:0 dropped:21 overruns:0 frame:0
          TX packets:83 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:26326 (25.7 KiB)  TX bytes:14359 (14.0 KiB)

```

[iotit.log.zip](https://github.com/xshellinc/iotit/files/826633/iotit.log.zip)

e-asphyx, 2017-03-18:
Might me the same problem: https://github.com/xshellinc/iotit/issues/25

makeupsomething, 2017-03-25:
This worked for me, but only after logging in once.

- Flashed device and set static ip to `192.168.10.167`
- Logged in over usb with `192.168.2.15`
- `ifconfig` showed wlan0 as `192.168.10.133`
- ssh with `192.168.10.133`
- Reboot device
- Try to ssh with `192.168.10.133` and cannot
- Logged in over usb with `192.168.2.15`
- `ifconfig` showed wlan0 as `192.168.10.167`

after that `192.168.10.167` is the ip address every time, which is the desired outcome

makeupsomething, 2017-04-18:
Il write about this quirk in the wiki and we can move this to ice box for now

kylycht, 2017-04-26:
ok, reboot post configuration did not work. but I found `correct(?)` way to set static IP address
it seems it is managed by `wpa_cli` daemon which runs `/etc/wpa_supplicant/wpa_cli-actions.sh`

```
if [ "$CMD" = "CONNECTED" ]; then
    kill_daemon udhcpc /var/run/udhcpc-$IFNAME.pid
    ifconfig $IFNAME 192.168.0.254 netmask 255.255.255.0 #new line
    route add default gw 192.168.0.1 #new line
    #disable dhcp
    #udhcpc -i $IFNAME -p /var/run/udhcpc-$IFNAME.pid -S
fi
```

I added new config and this seems to work

dbsparkle, 2017-05-08:
is this issue still valid? @makeupsomething

makeupsomething, 2017-05-08:
@dbsparkle testing it today

makeupsomething, 2017-05-09:
@kylycht Using latest version I still had to reboot manually before the device got the static ip address i set.

closed 2017-05-10 by makeupsomething