xshellinc/iotit #30: Password input problem when updating edison settings

makeupsomething, 2017-03-14:
When I want to update edisons settings I get asked to input the password. Even inputting correctly I get asked many times to do it. Im not sure if this problem is with iotit or edison os.

```
daryl-no-MacBook-Pro:iotit daryl$ ./iotit
   [1] raspberry-pi
   [2] edison
   [3] nano-pi
   [4] beaglebone
[?] Select device type: 2
[+] flashing edison
[?] Would you like to flash your device?  (y/yes OR n/no):n
   [1] Default
   [2] Enter IP
[?] Chose the edison's inteface to connect via usb: 1
   [1] awdl0
   [2] bridge0
   [3] en0
   [4] en1
   [5] en2
   [6] en4
   [7] fw0
   [8] gif0
   [9] lo0
   [10] p2p0
   [11] stf0
[?] Please chose correct interface: 6
[+] Enter Password: ********
[+] Enter Password: ******
[?] Would you like to assign static IP wlan address for your device? (y/yes OR n/no):n
The authenticity of host '192.168.2.15 (192.168.2.15)' can't be established.
ECDSA key fingerprint is SHA256:GFKKabbTbd7nj7Ut6mHyVKhTXY1WRtAhQzgnbLb4ViY.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '192.168.2.15' (ECDSA) to the list of known hosts.
root@192.168.2.15's password: 
Connection to 192.168.2.15 closed.
root@192.168.2.15's password: 
root@192.168.2.15's password: 
root@192.168.2.15's password: 
```

[iotit.log.zip](https://github.com/xshellinc/iotit/files/840766/iotit.log.zip)

larenso, 2017-03-22:
@makeupsomething it happens if you set a password for edison, thus each time it sends a command it requests your pass.

makeupsomething, 2017-03-22:
@larenso I guess then it is expected behaviour? Shall we close the issue?

larenso, 2017-03-22:
@makeupsomething I'm trying to enhance it

makeupsomething, 2017-04-25:
I cannot connect to the edison at all now
```
daryl-no-MacBook-Pro:iotit daryl$ ./iotit
   [1] raspberry-pi
   [2] edison
   [3] nano-pi
   [4] beaglebone
   [5] Custom
[?] Select device type: 2
[+] flashing edison
[?] Would you like to flash your device?  (y/yes OR n/no):n
[?] Please unplug your edison board. Press yes once unpluged?  (y/yes OR n/no):y
ssh: connect to host localhost port 2222: Connection refused
[-] Cannot find mounted Intel edison device, please try to re-mount it
[?] Press yes once mounted?  (y/yes OR n/no):y
ssh: connect to host localhost port 2222: Connection refused
[-] Cannot find mounted Intel edison device, please try to re-mount it
[?] Press yes once mounted?  (y/yes OR n/no):y
ssh: connect to host localhost port 2222: Connection refused
[-] Cannot find mounted Intel edison device, please try to re-mount it
[?] Press yes once mounted?  (y/yes OR n/no):y
ssh: connect to host localhost port 2222: Connection refused
[-] Cannot find mounted Intel edison device, please try to re-mount it
[?] Press yes once mounted?  (y/yes OR n/no):y
ssh: connect to host localhost port 2222: Connection refused
[-] Cannot find mounted Intel edison device, please try to re-mount it
[?] Press yes once mounted?  (y/yes OR n/no):
```

should it not be port 22?

kylycht, 2017-04-26:
@makeupsomething it looks like it is trying to connect to vbox (port 2222 mapped to 22 on vbox)
but I could not recreate this bug

makeupsomething, 2017-04-28:
@kylycht @bnuriddin 
About this error
```
[?] Select device type: 2
[+] flashing edison
[?] Would you like to flash your device?  (y/yes OR n/no):n
[?] Please unplug your edison board. Press yes once unpluged?  (y/yes OR n/no):y
ssh: connect to host localhost port 2222: Connection refused
 ```

Some extra info on this. It happens when only doing sevice setup (not flashing). It seems that when you pick that option the vbox does not start. When i started vbox manually and ran again I got same error but it was a little but different. This line in particualr `ash: /tmp/flashall.sh: not found`

```
ash: /tmp/flashall.sh: not found
[-] Cannot find mounted Intel edison device, please try to re-mount it
[?] Press yes once mounted?  (y/yes OR n/no):y
DDDDDD root localhost 2222
ash: /tmp/flashall.sh: not found
[-] Cannot find mounted Intel edison device, please try to re-mount it
[?] Press yes once mounted?  (y/yes OR n/no):^C
```

makeupsomething, 2017-04-28:
So i guess if vbox did not start correctly it makes sense that it would not be able to find mounted edison

kylycht, 2017-04-28:
@makeupsomething if you skip flashing, why it would attempt to invoke `flashall.sh` script. Something is wrong. I will check

kylycht, 2017-04-28:
ok, I think I recreated this issue. fixing

makeupsomething, 2017-05-08:
@kylycht This issue seems resolved now, as I can update the settings without problem but now when I try to flash all it does it update the settings also.

makeupsomething, 2017-05-10:
Still happens when updating settings but does not happen when flashing

closed 2017-05-10 by makeupsomething