xshellinc/iotit #40: Create submenus for raspberry devices and nano devices

makeupsomething, 2017-03-24:
See https://github.com/xshellinc/isaax-main/issues/136 for more info

```
➜  CLI ./iotit
   [1] Raspberry
   [2] Intel Edison
   [3] Nano Pi
   [4] Beaglebone
```
and inside there have sub choices for raspberry (or any other board we support in future)

```
Select raspberry Pi model to flash:
   [1] Raspberry Pi Model A, A+, B, B+, Zero, Zero W
   [2] Raspberry Pi 2 (based on Model B+)
   [3] Raspberry Pi 3
```

And inside nano pi sub menu, we can group some of these later
```
Select raspberry Pi model to flash:
   [1] NanoPi2
   [2] NanoPi 2 Fire
   [3] NanoPi M1 Plus
   [4] NanoPi M1
   [5] NanoPi M2
   [6] NanoPi M3
   [7] NanoPi NEO
   [8] NanoPi NEO Air
   [9] NanoPi S2
   [10] NanoPi a64
```

larenso, 2017-03-28:
should we clone all nano-pi images to our repo?

dbsparkle, 2017-03-28:
https://github.com/xshellinc/iotit/issues/40#issuecomment-289665854

No, definitely not. What for?

larenso, 2017-03-28:
Seems like some nano-pi devices have different images and our Nano-Pi Neo's image (which is currently used for all devices) doesn't work with Nano-pi M3

dbsparkle, 2017-03-29:
@larenso you didn't answer why we have to clone the images to our bistro instead of downloading from the original repository

larenso, 2017-03-29:
@dbsparkle a copy in our distro guarantees that the image is accessible, and we are safe from changes on the nano-pi side.
Otherwise we shall make a file with urls bindings which should be tracked and updated
for example currently image for nano-pi neo could be found here:
`http://download1851.mediafire.com/7xa5i1adz02g/iiabiw4r3r70at9/nanopi-neo-ubuntu-core-qte-sd4g-20170220.img.zip`

for m3:
`http://download1326.mediafire.com/v3245vadbblg/dts72ru5vzzzsr5/s5p6818-ubuntu-core-qte-sd4g-20170316.img.zip`

The roms are available from their wiki i.e. `http://wiki.friendlyarm.com/wiki/index.php/NanoPi_M3`
And currently I don't see any logic, except downloading it to our repo, or creating a file with device-url map

makeupsomething, 2017-03-29:
@larenso @dbsparkle This is just my opinion but I think we should use a device-url map file. The idea behind iotit being open source is that we don't want to be maintaining it too much. If somebody adds functionality for a new device type then all they need to do is add the download url to the file and then it should work. If we are hosing the images then there is extra work on our side to download and host every time a new OS is supported.

larenso, 2017-03-29:
@makeupsomething currently we have copies of nanopi, raspberry, beaglebone and edison
should we move them to the map-file too?

makeupsomething, 2017-03-29:
@larenso Yep, i would suggest to move them to the map too. We don't want to host anything. If we are worried about one host being un reliable then we can map one device to multiple sources.

closed 2017-04-11 by larenso