xshellinc/iotit #45: add option to flash with registration token

kylycht, 2017-03-28:
Add option where user can store registration token during init of the device
**NOTE: Token should be present in ONE of these options**

Order of token lookup:
1. Environment variable `ISAAX_TOKEN`
2. Read from file `~/.isaax_token`
3. Read from file `/etc/.isaax_token`

kylycht, 2017-03-28:
@larenso @dbsparkle maybe we should pack `isaax-agent` during `init`?

dbsparkle, 2017-03-28:
@kylycht 

1. Does it mean that we will have files `~/.isaax_token` and `/etc/.isaax_token` ?
2. this will allow us to upload isaax-agent, isaax_token during `init` - what is `init` here?

kylycht, 2017-03-28:
@dbsparkle it is not required. it is ONE of the options. Agent will lookup for environment variable in that order. In some cases there might be no `~`(home dir for user), 
in that case it can be placed in `/etc/.isaax_token`

The order is: 
1. Lookup envar
2. If envar does not exist, lookup in home dir
3. If home dir does not exist, lookup in `/etc/`

> NOTE: Token should be present in ONE of these options

kylycht, 2017-03-28:
So there might be a prompt something like:

```
[+] Would you like to upload additional files?(y/n):
#if user selects yes
[+] Please enter full path to your file: /tmp/registration_token_file
[+] Please enter destination dir: /etc/.isaax_token
```

dbsparkle, 2017-03-28:
Maybe it should be a common file, something like `etc/.isaax_config`?

kylycht, 2017-03-28:
@dbsparkle we can name it `.isaax_config`, but it is going to contain only a token

dbsparkle, 2017-03-29:
https://github.com/xshellinc/iotit/issues/45#issuecomment-289799318

Keep in mind footprint problem @kylycht

kylycht, 2017-03-29:
@dbsparkle yep, need to review `isaax-agent` config file and remove what can be removed

larenso, 2017-04-17:
can be created smth like `./iotit/sd-copy/`

and all the content of the `sd-copy` will be copied to `sd_card` recursively replacing files
i.e. `./iotit/sd-copy/etc/dhcpcd.conf` copy and replace if exists `/etc/dhcpcd.conf`

dbsparkle, 2017-08-07:
@kylycht @tectiv3 I think we should close this issues as an abandoned one

closed 2017-08-07 by tectiv3