xshellinc/iotit #47: Add `Custom` flash option

makeupsomething, 2017-03-28:
This is a **discussion** to see how we would do this, and if we should do it or leave it open to community.
Add an option to allow the user to flash any os image they want from their drive. 

```
➜  CLI ./iotit
   [1] Raspberry
   [2] Intel Edison
   [3] Nano Pi
   [4] Beaglebone
   [5] Custom
```

Like what Etcher does, this option will be for users who already have an idea of what they are doing and trying to install. We ill provide guidance but no guarantees that the flashed image will work for their device/arch.

larenso, 2017-04-17:
Changed workflow, now, when user starts iotit, it creates `.iotit/mapping.json`
this file can be used in order to add other images in the simple format:

```
[
  Name: device name,
  Images: [
    {
      Url: url to download a zip/tar/xv image
    }
  ]
]
```

dbsparkle, 2017-04-17:
@larenso please reflect this: https://github.com/xshellinc/iotit/issues/47#issuecomment-294423481
in README file.

makeupsomething, 2017-04-18:
@bnuriddin `mapping.json` does not seem to get created on my machine. Looked in `/tmp/` and the project root folder but not in either.

makeupsomething, 2017-04-18:
Also, along with the above method of the adding the URL you want to download from into the `mapping.json` file can we not also have the user input it directly into the cli if they select custom. For example:
```
[5] Custom
[?] Select device type: 5
[+] flashing Custom
[-] Error:  No custom boards are available
[+] Please enter image url
```
Then it will download/extract and intall from the url. 

Is this possible? @larenso @bnuriddin

larenso, 2017-04-18:
yep

larenso, 2017-04-20:
@makeupsomething pls check in `bugfix/unzip-fix` branch

makeupsomething, 2017-04-21:
@larenso has the method of flashing by `mapper.json` file been abandoned now? It still doesnt seem to get created on my machine. 

Flashing by passing URL in directly works though

larenso, 2017-04-21:
nope, it is created inside `.iotit`. It is useful as a configuration file

makeupsomething, 2017-04-22:
@larenso At what point is it created? As soon as you run iotit? I had to create it manually to test.

larenso, 2017-04-25:
Yep at the first start it creates a folder `.iotit` and `mapping.json`, you can also run `iotit help` to create this file

makeupsomething, 2017-04-25:
@larenso Thanks, got it now

tectiv3, 2017-05-17:
Please checkout `bugfix/custom-boards` branch

makeupsomething, 2017-05-18:
@tectiv3

The flashing by filepath works well.

But when i flash by url im getting this error
```
daryl-no-MacBook-Pro:iotit daryl$ ./iotit
Log location: /var/folders/wy/bxm050ps2bl8j_lc4g3n0dfc0000gn/T/iotit.log
   [1] raspberry-pi
   [2] edison
   [3] nano-pi
   [4] beaglebone
   [5] custom board
[?] Select device type: 5
[+] Flashing custom board
[-] No custom boards are available
[?] Please enter image location URL or filepath: http://vx2-downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-04-10/2017-04-10-raspbian-jessie-lite.zip
[+] Checking virtual machine
[+] No problem!
[+] Checking running virtual machine
[?] iotit-box (default) is running, would like you stop this virtual machine? (y/yes, n/no): y
   [1] Use default vbox preset
   [2] Create a new vbox preset
[?] Please select an option: 1
[+] Selected virtual machine
	Name - iotit-box
	Description - 
[+] Starting: \ success

[+] Starting download custom board
[+] Download 2017-04-10-raspbian-jessie-lite.zip 293.53 MB / 293.53 MB  100.00% 0s                                                                                                   
[+] Extracting  
[-]  ash: syntax error: unexpected word (expecting ")")

[-] Error:  Cannot find a mounting point
```

Here is the logfile

[iotit.log.zip](https://github.com/xshellinc/iotit/files/1009582/iotit.log.zip)

tectiv3, 2017-05-18:
Fixed

closed 2017-05-18 by makeupsomething