xshellinc/iotit #48: Double ip on raspberry pi

makeupsomething, 2017-03-28:


makeupsomething, 2017-03-28:
```
eth0      Link encap:Ethernet  HWaddr b8:27:eb:8a:86:4a  
          inet6 addr: fe80::3f6:c7fd:3cfd:8ced/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr 88:57:ee:99:2d:84  
          inet addr:192.168.10.170  Bcast:192.168.10.255  Mask:255.255.255.0
          inet6 addr: 2408:210:2ce3:800:748c:f4d4:ec34:ee39/64 Scope:Global
          inet6 addr: fe80::e72a:5da4:3fad:674c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:143938 errors:0 dropped:13831 overruns:0 frame:0
          TX packets:31337 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:137998977 (131.6 MiB)  TX bytes:3554914 (3.3 MiB)

```

makeupsomething, 2017-03-28:
```
{"timestamp":"2017-03-28T10:21:36.133547099Z","payload":{"device":{"hostname":"raspberrypi","interfaces":[{"ip":"127.0.0.1","name":"lo","mac":""},{"ip":"192.168.10.170","name":"wlan0","mac":"88:57:ee:99:2d:84"},{"ip":"192.168.10.143","name":"wlan0","mac":"88:57:ee:99:2d:84"}],"timezone":"UTC","date":"�","os":"linux","arch":"arm"}}}
```

makeupsomething, 2017-03-28:
## /etc/network/interfaces
```
cat  /etc/network/interfaces
source-directory /etc/network/interfaces.d

auto lo
iface lo inet loopback

iface eth0 inet manual

allow-hotplug wlan0
iface wlan0 inet static
address 192.168.10.170
netmask 255.255.255.0
network 192.168.10.0
gateway 192.168.10.1
dns-nameservers 192.168.10.1
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```

bnuriddin, 2017-03-28:
@makeupsomething can you run this `ip address show wlan0` and show us output please

makeupsomething, 2017-03-28:
```
ip address show wlan0
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 88:57:ee:99:2d:84 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.170/24 brd 192.168.10.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet 192.168.10.143/24 brd 192.168.10.255 scope global secondary wlan0
       valid_lft forever preferred_lft forever
    inet6 2408:210:2ce3:800:748c:f4d4:ec34:ee39/64 scope global noprefixroute dynamic 
       valid_lft 2591977sec preferred_lft 604777sec
    inet6 fe80::e72a:5da4:3fad:674c/64 scope link 
       valid_lft forever preferred_lft forever
```

bnuriddin, 2017-03-28:
hmm, could you check if `/etc/dhcpcd.conf` exists and put it here if it does. Looks like there is another configuration file on system

bnuriddin, 2017-03-28:
@larenso have you been playing with dhcp configuration, when setting Google DNS as secondary IP? Do you install or modify dhcpd service?

larenso, 2017-03-28:
@bnuriddin I changed only `etc/dhcp/dhclient.conf` config

larenso, 2017-03-28:
`append domain-name-servers 8.8.8.8, 8.8.4.4;`

bnuriddin, 2017-03-28:
@larenso let's get config from @makeupsomething  but looks like this activates `dhcpd` service

larenso, 2017-03-28:
https://github.com/xshellinc/tools/commit/c86c8dc2e69b8696b3449e5f01e51eebd4b59c08

larenso, 2017-03-28:
@bnuriddin also this was changed

bnuriddin, 2017-03-28:
so here is what was causing it:
on raspbian network configuration made a little different. dhcpd service is enable and running always nad if we set static IP in `interfaces` file it will ignore changes and connect to dhcp anyways.

Correct way for raspbian is to leave `interfaces` configuration for settting static IP and do it in `/etc/dhcpcd.conf`
Example
```
     static value
             Configures a static value.  If you set ip_address then dhcpcd
             will not attempt to obtain a lease and just use the value for the
             address with an infinite lease time.

             Here is an example which configures a static address, routes and
             dns.
                   interface eth0
                   static ip_address=192.168.0.10/24
                   static routers=192.168.0.1
                   static domain_name_servers=192.168.0.1
```
See more at: http://www.daemon-systems.org/man/dhcpcd.conf.5.html

makeupsomething, 2017-03-29:
@bnuriddin 

```
cat /etc/dhcpcd.conf
# A sample configuration for dhcpcd.
# See dhcpcd.conf(5) for details.

# Allow users of this group to interact with dhcpcd via the control socket.
#controlgroup wheel

# Inform the DHCP server of our hostname for DDNS.
hostname

# Use the hardware address of the interface for the Client ID.
clientid
# or
# Use the same DUID + IAID as set in DHCPv6 for DHCPv4 ClientID as per RFC4361.
#duid

# Persist interface configuration when dhcpcd exits.
persistent

# Rapid commit support.
# Safe to enable by default because it requires the equivalent option set
# on the server to actually work.
option rapid_commit

# A list of options to request from the DHCP server.
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
# Most distributions have NTP support.
option ntp_servers
# Respect the network MTU.
# Some interface drivers reset when changing the MTU so disabled by default.
#option interface_mtu

# A ServerID is required by RFC2131.
require dhcp_server_identifier

# Generate Stable Private IPv6 Addresses instead of hardware based ones
slaac private

# A hook script is provided to lookup the hostname if not set by the DHCP
# server, but it should not be run by default.
nohook lookup-hostname
```

larenso, 2017-04-17:
Now writing configuration into /etc/dhcpcd.conf
Moreover a newer image of raspbian has the ssh turned off. I've added an instruction to enable it via
renaming `/etc/rc2.d/K*ssh` into `S02ssh`

larenso, 2017-04-17:
Also created a custom configurator, so that we sdFlasher can use default configuration workflow or override or customize it: in the particular raspberry pi case: workflow changes configuring pi interfaces and enabling ssh

makeupsomething, 2017-04-18:
Tested on same device that had the problem before. Could not reproduce.

closed 2017-04-18 by makeupsomething