xshellinc/iotit #59: Are we supporting/testing on windows?

makeupsomething, 2017-04-17:
Just thinking about this today. 
Are any of you guys running windows? 

@bnuriddin @e-asphyx @larenso

bnuriddin, 2017-04-17:
I don't

tectiv3, 2017-05-05:
So, first problem arises:
```
[-] Error:  [-] Could not find virtualbox, please install virtualbox
```
It can't find `VBoxManage` with `which`. Cause it's not in the `PATH`.
It's in `C:\Program Files\Oracle\VirtualBox\VBoxManage.exe` on windows.

So, easy fix would be to check for windows platform and run it directly from `Program Files`, hardcode it.

Or we could ask user to add `C:\Program Files\Oracle\VirtualBox\` to their PATH.

Or we can detect windows, try to run it from program files and fallback to asking user about fixing their PATH.

Which path should be choose?

dbsparkle, 2017-05-05:
Well, User should always be aware of changes/updates of PATHs in his system, so I think we should display a prompt recommending him to set it the way we want...

tectiv3, 2017-05-05:
Ok, now the hard part:
`package workstation` contains only:
```
// @todo add windows methods
```
So there is no windows sd card burning support at all. Everything else I've fixed (virtual box, paths and text formatting issues)
I will start working on adding burning, monday I think.

tectiv3, 2017-05-09:
```
PS Z:\src\github.com\xshellinc\iotit> .\iotit.exe --dev=raspberry-pi
Log location: C:\Users\IEUser\AppData\Local\Temp\iotit.log
[+] list drives!
   [1] [15GB] JetFlash Transcend 16GB USB Device - \\?\Device\Harddisk1\Partition0
[?] Select disk to format:
```
I can't always get a drive letter, is this approach ok? Size and model will be printed on the left side and physical address on the right.

makeupsomething, 2017-05-09:
@tectiv3 Yes I think this is ok, in the case you can't get the letter. The user should be able to tell based of the name. At what rate can you not get the drive letter 1 in 10? 1in 5? If its quite common we can add a warning to the user.

tectiv3, 2017-05-09:
Well, it's not that I can't get the letter, it's hard to map it to the physical device. Windows API either lists physical drives or logical. And only logical have letters. And to be able to burn image – we need physical drives and letters aren't useful to us.

makeupsomething, 2017-05-09:
Yep I agree. Anyway user should be able to determine which drive is correct so its ok to drop letters.

tectiv3, 2017-05-09:
Ok, guys, I need a second opinion here.
Can someone with access to `windows` machine or properly setup VM check my code for sanity.
What am I doing:
-list diskdrives using `wmic`
-use `dd-win` to put image on selected drive
This was supposed to be nice and easy. But it's not. Windows is sometimes refusing me to write on that drive.
I get `5 Access denied` error. And from what I googled it's because something is using the device. I found no way to figure out what it is to stop this from happening. Using `mountvol` to dismount disk only make things worse, I get `device isn't ready` error.

But I've got a solution. As ugly as windows is.
I've added one more command to iotit – `clean`. What it does is run `diskpart` to clean target disk partitions and create a single fat32 one.
After that, simple reconnecting SD card or flash drive allows `iotit` to successfully burn the image to disk.
I don't know, but either I am missing something or it's just a regular windows way that some things almost work, but not good enough.
So, I need someone else to look into it.

Apart from this issue, it works :)

makeupsomething, 2017-05-10:
@tectiv3 I think I am the only person who uses a windows machine at home. So il test this out!

tectiv3, 2017-05-10:
Now it should work fine.
Burning process still a bit messy.
If you try to burn SD card or Flash drive that contains only one partition - then it would be fine.
But if your drive has more than one partition, then you will need to run `iotit clean`, reconnect your drive/card and try again (you can launch another admin PowerShell process to clean drive).

hm, it got me thinking, why ask user to run that command separately when I can ask that question right inside the app...

tectiv3, 2017-05-11:
Now it needs testing.
I've tested it with raspi (nano-pi and BB process essentially is the same) and Edison I can test in a few weeks when it arrives.

tectiv3, 2017-05-12:
Closing this. Please submit a new issue in case bugs on `windows` being found

closed 2017-05-12 by tectiv3