xshellinc/iotit #60: Unzip error

makeupsomething, 2017-04-18:
I got inflate error while flashing raspberry pi 2

```
   [1] Use default vbox preset
   [2] Create a new vbox preset
[?] Please select an option: 1
[+] Selected virtual machine 
	[Name] - iotit-box
	[Description] - 
[+] starting: / success

[+] Starting download  raspberry-pi
[+] Download 2017-03-02-raspbian-jessie-lite.zip 648.60 KB / 648.60 KB  100.00% 0s                                                                                                   
[+] Uploading 2017-03-02-raspbian-jessie-lite.zip to virtual machine
[+] Extracting 2017-03-02-raspbian-jessie-lite.zip 
[-]  unzip: unexpected end of file
unzip: inflate error

[?] Would you like to configure your board? (y/yes OR n/no):y
[+] Default language:  en_US.UTF-8
[?] Change default language? (y/yes OR n/no):y
```

larenso, 2017-04-18:
Seens like bad archive, `648.60 KB` is pretty small, program should exit and probably recheck the archive name

dbsparkle, 2017-04-20:
@makeupsomething @larenso whats the status here?

larenso, 2017-04-20:
@dbsparkle added content length check, and fixed error status:
```
unzip: unexpected end of file
unzip: inflate error
```
now stop the tree execution and exits, no configurations is asked

makeupsomething, 2017-04-21:
It was already quite a rare bug as we need a bad image to recreate. I will put in icebox and consider done unless it comes up again.

closed 2017-07-26 by dbsparkle