xshellinc/iotit #61: LANGUAGE & LC_ALL left unset

dbsparkle, 2017-04-18:
@kylycht commented on [Fri Jan 06 2017](https://github.com/xshellinc/isaax-cli/issues/95)

@dbsparkle commented on [Thu Jan 05 2017](https://github.com/xshellinc/isaax-agent/issues/50)

```
warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
```

Shall we set this during `device init` ?

---

@kylycht commented on [Fri Jan 06 2017](https://github.com/xshellinc/isaax-agent/issues/50#issuecomment-270820571)

@dbsparkle if it is feasible, we should I guess

---

@larenso commented on [Fri Jan 27 2017](https://github.com/xshellinc/isaax-cli/issues/95#issuecomment-275591512)

@dbsparkle @makeupsomething 
according to this:
https://www.gnu.org/software/gettext/manual/html_node/Locale-Environment-Variables.html
https://www.gnu.org/software/gettext/manual/html_node/The-LANGUAGE-variable.html#The-LANGUAGE-variable
`some Swedish users who would rather read translations in German than English for when Swedish is not available, set LANGUAGE to ‘sv:de’ while leaving LANG to ‘sv_SE’.`

I currently implemented that LANGUAGE = LC_ALL = LANG
or should we add an option to set a global language (LC_ALL, LANG) and your language(LANGUAGE)?

---

@dbsparkle commented on [Fri Jan 27 2017](https://github.com/xshellinc/isaax-cli/issues/95#issuecomment-275592229)

LC_ALL, LANG is ok

---

@larenso commented on [Fri Jan 27 2017](https://github.com/xshellinc/isaax-cli/issues/95#issuecomment-275628692)

@makeupsomething pls check in https://github.com/xshellinc/isaax-cli/tree/feature/device-ip-check

---

@larenso commented on [Fri Feb 17 2017](https://github.com/xshellinc/isaax-cli/issues/95#issuecomment-280629710)

Noticed some problems, during rc.local script uses locale set by a user, but after that it suddenly reverts to defaults.
probably /etc/locale.conf - is not suited enough for that, trying to workaround this by setting local user variables and /etc/environment

tectiv3, 2017-05-16:
The problem is in macOS terminal.
The easiest fix would be to comment out`SendEnv LANG LC_*` in mac's `/etc/ssh_config`

tectiv3, 2017-05-18:
Example:
```sh
pi@raspberrypi.local's password:

-bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
-bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
-bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
pi@raspberrypi:~ $ 
Connection to raspberrypi.local closed.
```
##### Disable `SendEnv LANG LC_*`
```sh
$ sudo mate /etc/ssh/ssh_config
```
```
$ ssh pi@raspberrypi.local
pi@raspberrypi.local's password:
Last login: Thu May 18 07:20:04 2017 from 192.168.0.15
pi@raspberrypi:~ $
```

closed 2017-05-18 by tectiv3