xshellinc/iotit #64: Error extracting virtual machine

makeupsomething, 2017-04-24:
I get this error when iotit extracts VM from `~/.iotit/` folder 

```
[+] Checking virtual machine
[+] Unzipping /Users/daryl/.iotit/virtualbox/0.0.3/iotit-box.zip to /Users/daryl/VirtualBox VMs
[+] Unzipping iotit-box.zip   0 B / 136.00 MB    0.00%                          [-]  open /Users/daryl/VirtualBox VMs/iotit-box/iotit-box.vbox: permission denied
[-] Error:  open /Users/daryl/VirtualBox VMs/iotit-box/iotit-box.vbox: permission denied
[-] Exiting ... 
```
Extracting manually works.

[iotit.log.zip](https://github.com/xshellinc/iotit/files/950887/iotit.log.zip)

larenso, 2017-04-25:
could you check permissions:
```
ls -a /Users/daryl/VirtualBox VMs/
ls -a /Users/daryl/VirtualBox VMs/iotit-box/
```

makeupsomething, 2017-04-25:
@larenso 
```            
daryl-no-MacBook-Pro:isaax-agent daryl$ ls -a /Users/daryl/VirtualBox 
.DS_Store           CHANGELOG.md        app/                dist/               isaax-agent         manager_test.go     updater.go          worker/             
.buildkite/         Dockerfile          bin/                docker-compose.yml  logging/            process/            updater_test.go     
.git/               Makefile            broker/             installer.go        main.go             registrer.go        upgrader/           
.gitignore          README.md           coverage.out        installer_test.go   manager.go          registrer_test.go   util/   
```

```
daryl-no-MacBook-Pro:isaax-agent daryl$ ls -l /Users/daryl/VirtualBox\ VMs
total 0
drwxr-xr-x  6 daryl  staff  204 Apr 24 17:29 iotit-box
```

```            
daryl-no-MacBook-Pro:isaax-agent daryl$ ls -a /Users/daryl/VirtualBox\ VMs/iotit-box/
.			..			Logs			iotit-box.vbox		iotit-box.vbox-prev	iotit-box.vdi
```
```
daryl-no-MacBook-Pro:isaax-agent daryl$ ls -l /Users/daryl/VirtualBox\ VMs/iotit-box/
total 2738192
drwx------  5 daryl  staff         170 Apr 24 17:14 Logs
-rw-------  1 daryl  staff        4024 Apr 24 17:29 iotit-box.vbox
-rw-------  1 daryl  staff        4024 Apr 24 17:14 iotit-box.vbox-prev
-rw-r--r--  1 daryl  staff  1401946112 Apr 24 17:29 iotit-box.vdi
```

kylycht, 2017-04-26:
Have fixed the issue above, now getting: 
```
[?] Select device type: 2
[+] flashing edison
[?] Would you like to flash your device?  (y/yes OR n/no):y
[+] Checking virtual machine
[+] No problem!
[-] Error:  machine does not exist
[-] Exiting ... 
```

makeupsomething, 2017-04-28:
We had no estimate points for this. @kylycht said he spent about 10 hours on it so im setting 20 pts.

makeupsomething, 2017-04-28:
Oh, my mistake, we did have 3 hours on it...but as it became multiple problems it took longer.

closed 2017-04-28 by makeupsomething