xshellinc/iotit #65: Let's add ability to write into Raspi boot partition

tectiv3, 2017-05-01:
As I see this one, it requires modification in several places.

Here we need to mount both image partitions to different mounting points.
#### rasppi.go:28
```go
func (d *raspberryPi) Configure() error 
```
And here, I think, `MountImg` should support custom `MountDir`
#### sd_flasher.go:35
```go
func (d *sdFlasher) MountImg(loopMount string) error
```
Currently `MountImg` only wants loop device number and mounts it into `/tmp/isaax-sd/` constant

tectiv3, 2017-05-11:
Any ideas? Suggestions? Please

makeupsomething, 2017-05-11:
> Any ideas? Suggestions? Please
@yutashi @kylycht @bnuriddin

bnuriddin, 2017-05-11:
When user chooses sd-card drive name I guess we can determine on our own where is `/boot` and `/` and other partitions

> Here we need to mount both image partitions to different mounting points.
> And here, I think, MountImg should support custom MountDir
@tectiv3 already stated what need to be done.
custom mountdirs and dirs in `/tmp/isaax-sd/` according to these moundirs.

tectiv3, 2017-05-11:
Well, it's not about physical device, it's about an image. We need to write to it before flashing, on a configuration stage.

Currently we are attaching it through a loop device using hardcoded loop0p2, meaning it's only the second partition.
10 days ago I knew significantly less about the process hence the question :)

bnuriddin, 2017-05-11:
> Well, it's not about physical device, it's about an image
got them mixed, my bad.

closed 2017-05-12 by tectiv3