xshellinc/iotit #74: Enable ssh not working for raspberry

makeupsomething, 2017-05-18:
This did not seem to enable ssh and i had to enable it manually myself after flashing.

```
[?] Would you like to enable SSH server? (y/yes, n/no): y
```

tectiv3, 2017-05-18:
This is weird, I tested it on my pi. I'll check it again.

tectiv3, 2017-05-18:
Fixed. Surprisingly `losetup -D` does not detach all loop devices :)

tectiv3, 2017-05-18:
```
[+] Stopping virtual machine
****************************************************************************************************
*		 SD CARD READY!
*		 PLEASE INSERT YOUR SD CARD TO YOUR raspberry-pi
*		 IF YOU HAVE NOT SET UP THE USB WIFI, PLEASE CONNECT TO ETHERNET
*		 SSH USERNAME:pi PASSWORD:raspberry
****************************************************************************************************
$ ssh pi@raspberrypi.local
The authenticity of host 'raspberrypi.local (192.168.0.9)' can't be established.
ECDSA key fingerprint is SHA256:P9s4+VY6xWeZqWr3KqXdMgecz/dzxMhwGlC2lRyCxi0.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'raspberrypi.local,192.168.0.9' (ECDSA) to the list of known hosts.
pi@raspberrypi.local's password:
```

closed 2017-05-18 by makeupsomething