xshellinc/iotit #75: SSH disabled for raspberry pi after flash

bnuriddin, 2017-05-18:
appears on latest iotit, no matter if ssh enabled or disabled during `device init` configuration
```
[?] Would you like to enable SSH server? (y/yes, n/no): n
```
```
[?] Would you like to enable SSH server? (y/yes, n/no): y
```
```
➜  ~ sudo nmap 192.168.43.155

Starting Nmap 7.40 ( https://nmap.org ) at 2017-05-18 11:18 +06
Nmap scan report for 192.168.43.155
Host is up (0.056s latency).
All 1000 scanned ports on 192.168.43.155 are closed
MAC Address: F4:F2:6D:0F:5B:FF (Tp-link Technologies)

Nmap done: 1 IP address (1 host up) scanned in 3.77 seconds
```
```
➜  isaax-cli git:(master) ssh pi@192.168.43.155
ssh: connect to host 192.168.43.155 port 22: Connection refused
```

makeupsomething, 2017-05-18:
This is a duplicate, see #74

makeupsomething, 2017-05-18:
Also see

https://www.raspberrypi.org/blog/a-security-update-for-raspbian-pixel/

closed 2017-05-18 by makeupsomething