xshellinc/iotit #77: How can I confirm the correct mounting point?

yutashi, 2017-05-29:
version: iotit-0.2.1  
platform: mac OS

How can I confirm the correct mounting point to answer following question?
```
[+] Uploading 2017-04-10-raspbian-jessie-lite.img to virtual machine
   [1] loop0p1
   [2] loop0p2
[?] Please select the correct mounting point:
```

makeupsomething, 2017-05-29:
@tectiv3 Is there a way to make it more clear to the user which mounting point to use?

dbsparkle, 2017-05-29:
@makeupsomething @yutashi 
Read about loop here:  https://en.m.wikipedia.org/wiki/Loop_device

dbsparkle, 2017-05-29:
Simply think that loops are your partitions

makeupsomething, 2017-05-29:
> Simply think that loops are your partitions

We need a way to make that clear to the user when using iotit then, not sure how though. Unless theres some heuristic we can use to make a best guess at which one they should choose. If they use the default virtual machine its usually the second one.

dbsparkle, 2017-05-29:
@yutashi @makeupsomething TYI:

http://man7.org/linux/man-pages/man8/mount.8.html#THE_LOOP DEVICE

tectiv3, 2017-05-29:
To be honest, I don't think we even need this dialog. It's confusing for the user and not much of a help for us.
We can either mount every loop device we find inside chosen image and then figure out which one is linux root partition ourselves or always mount the second one because it would be a raspberry pi image anyway.

makeupsomething, 2017-05-29:
@tectiv3 If we can get rid of it then we should. It is not much use to the user.

dbsparkle, 2017-05-29:
Still it's worth mentioning about in documents

tectiv3, 2017-05-29:
@dbsparkle mentioning what? that we will automatically detect root partition?

dbsparkle, 2017-05-29:
Mentioning "how we detect the partitions"

closed 2017-06-01 by makeupsomething