xshellinc/iotit #81: Edit flashing finished message

makeupsomething, 2017-06-06:
Currently the message is a bit agressive.

```
****************************************************************************************************
*		 SD CARD READY!
*		 PLEASE INSERT YOUR SD CARD TO YOUR beaglebone
*		 IF YOU HAVE NOT SET UP THE USB WIFI, PLEASE CONNECT TO ETHERNET
*		 SSH USERNAME:ubuntu PASSWORD:temppwd
****************************************************************************************************
```

We should make it a bit more friendly. Lets include the `xshell` logo, or a new `iotit` logo and also a link to our helpdesk. 

```
   .                 .           .              .            .         .   .......        .                .            
    .              ..           .               .            .         .                  .                .            
     .            ..           .                .            .         .                  .                .            
      .          ..           .                 .            .         .                  .                .            
      ..        ..           .                  .            .         .                  .                .            
        .      ..          ..                   .            .         .                  .                .            
         .    ..          ..                    .            .         .                  .                .            
          .  .           ..                     .            .         .                  .                .            
                              .......           .  ........  .         .   .......        .                .            
          .  .                          .       .            .         .                  .                .            
         .    ..                       .        .            .         .                  .                .            
        .      ..                     .         .            .         .                  .                .            
      ..        ..                  ..          .            .         .                  .                .            
      .          ..                ..           .            .         .                  .                .            
     .            ..              ..            .            .         .                  .                .            
    .              ..            ..             .            .         .                  .                .            
   .                 .          ..              .            .         .   .......        .  .......       .   .......  
Finished flashing device!
User: ubuntu
Password: temppwd
If you have any issues or suggestions please see our help page at https://help.isaax.io/hc/ja or tweet us @isaax_iot
```

makeupsomething, 2017-06-06:
@dbsparkle @tectiv3 @SmevSmevson What do you guys think?

tectiv3, 2017-06-06:
Definitely need a friendlier message!

makeupsomething, 2017-06-13:
```
 ...                      .................    ..                
 ...                      .................   ....    ...        
 ...                             ....                 ...        
 ...          .....              ....                 ...        
 ...       ...........           ....         ...     .......... 
 ...      ...       ...          ....         ...     ...        
 ...     ...         ...         ....         ...     ...        
 ...     ...         ...         ....         ...     ...        
 ...     ...         ...         ....         ...     ...        
 ...     ....       ....         ....         ...      ...       
 ...      .....   .....          ....         ...      ....   .. 
 ...         .......             ....         ...        ....... 
```

makeupsomething, 2017-06-13:
Added new message

```
		 ...                      .................    ..                
		 ...                      .................   ....    ...        
		 ...                             ....                 ...        
		 ...          .....              ....                 ...        
		 ...       ...........           ....         ...     .......... 
		 ...      ...       ...          ....         ...     ...        
		 ...     ...         ...         ....         ...     ...        
		 ...     ...         ...         ....         ...     ...        
		 ...     ...         ...         ....         ...     ...        
		 ...     ....       ....         ....         ...      ...       
		 ...      .....   .....          ....         ...      ....   .. 
		 ...         .......             ....         ...        ....... 

		 Flashing Complete!
		 Please insert your sd card into your raspberry-pi
		 ssh to your board with the following credentials
		 ssh username:pi password:raspberry
		 If you have any issues or suggestions please see our help page at https://help.isaax.io/hc/ja or tweet us @isaax_iot

```

closed 2017-06-20 by makeupsomething