xshellinc/iotit #89: Add as package to Brew

dbsparkle, 2017-06-25:
See more here: https://github.com/Homebrew/brew

tectiv3, 2017-07-06:
```
$ brew tap xshellinc/iotit
==> Tapping xshellinc/iotit
Cloning into '/usr/local/Homebrew/Library/Taps/xshellinc/homebrew-iotit'...
remote: Counting objects: 5, done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 5 (delta 0), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (5/5), done.
Tapped 1 formula (30 files, 36.8KB)
[~/go/src/github.com/xshellinc/iotit]
$ brew install iotit --HEAD
==> Installing iotit from xshellinc/iotit
==> Cloning https://github.com/xshellinc/iotit.git
Updating ~/Library/Caches/Homebrew/iotit--git
==> Checking out branch develop
==> go build -ldflags -s -w -X main.version=HEAD-042df5e -o /usr/local/Cellar/iotit/HEAD-042df5e/bin/iotit
🍺  /usr/local/Cellar/iotit/HEAD-042df5e: 3 files, 7.3MB, built in 5 seconds
```

closed 2017-07-06 by tectiv3

tectiv3, 2017-07-06:
Bottles (bin packages) example:
https://github.com/goochjj/homebrew-custom/blob/master/bash.rb
https://github.com/goochjj/homebrew-custom/blob/master/update_bash_bottle.sh