xshellinc/iotit #92: Cannot change locale

makeupsomething, 2017-07-27:
When flash and set a new locale using iotit this error appears on raspberry pi 2 when booting up.

`-bash: warning: setlocale: LC_ALL: cannot change locale (uk_UA.utf8)
`

dbsparkle, 2017-07-27:
@makeupsomething uk_UA.utf8 is Ukraine locale...

tectiv3, 2017-07-27:
yeah, you've set it manually at configuration step?

makeupsomething, 2017-07-27:
@tectiv3 yep

tectiv3, 2017-07-27:
well, it means you need to run `sudo locale-gen uk_UA` and `sudo dpkg-reconfigure locales`
and @dbsparkle is right it's a ukrainian locale

dbsparkle, 2017-07-27:
I am wondering why Ukrainian locale has been set as default? :)

makeupsomething, 2017-07-27:
@tectiv3 Why do I have to run it? Should it not have been set by iotit?

Why is everyone mentioning that it is Ukraine? :)

makeupsomething, 2017-07-27:
@dbsparkle not by default, i set it during configuration

tectiv3, 2017-07-27:
@dbsparkle don't look at me I don't use ukrainian locale:)

tectiv3, 2017-07-27:
@makeupsomething well, this is needs to be run on an actual system. iotit only prepares image.

makeupsomething, 2017-07-31:
> ofc we can make some sort of startup script that run only on a first boot

Not necessary i think. Closing this issue.

closed 2017-07-31 by makeupsomething