xshellinc/iotit #94: Error updating iotit

makeupsomething, 2017-07-27:
When i run `iotit update` I get this error. It is latest version on develop branch.

```
iotit update
[+] Current os:  darwin amd64
[-] Error: strconv.Atoi: parsing "": invalid syntax
```

tectiv3, 2017-07-27:
This one is tricky. Since we moved to github releases and goreleaser we are no longer update
http://iotit.s3-website-us-west-2.amazonaws.com/version.json

I need to find a way to replicate it's functionality with github.

dbsparkle, 2017-08-07:
@tectiv3 what about this issue?

tectiv3, 2017-08-07:
well, I was planning on handling it this week. Probably even today.
It's disabled atm so it won't affect the release and since we are distributing iotit through package managers (brew and linux native packages) I guess it's not a big problem.

closed 2018-01-21 by tectiv3