xshellinc/iotit #97: Support Asus Tinker board

makeupsomething, 2017-08-29:
We want to add tinker board as a default supported board in iotit. This is to be done in September.

makeupsomething, 2017-08-29:
When flashing the android OS image as a custom board I got this error.

```
[?] Please provide image URL or path: /Users/daryl/Documents/20170511-tinker-board-android-marshmallow-userdebug-v13.11.0.4.img 
[+] Checking virtual machine
[?] iotit-box (default) is running, would like you stop this virtual machine? (y/yes, n/no): y
   [1] Use default vbox preset
   [2] Create a new vbox preset
[?] Please select an option: 1
[+] Selected virtual machine
    Name - iotit-box
    Description - 
[+] Starting: \ 
[+] Using local image file for  custom board
[+] Uploading 20170511-tinker-board-android-marshmallow-userdebug-v13.11.0.4.img to virtual machine
[-] Error:  Cannot find a mounting point
Cannot find a mounting point
github.com/xshellinc/iotit/device.(*sdFlasher).MountImg
    /Users/daryl/Documents/goProjects/src/github.com/xshellinc/iotit/device/sd_flasher.go:53
github.com/xshellinc/iotit/device.(*sdFlasher).Configure
    /Users/daryl/Documents/goProjects/src/github.com/xshellinc/iotit/device/sd_flasher.go:166
github.com/xshellinc/iotit/device.(*sdFlasher).Flash
    /Users/daryl/Documents/goProjects/src/github.com/xshellinc/iotit/device/sd_flasher.go:193
main.main.func1
    /Users/daryl/Documents/goProjects/src/github.com/xshellinc/iotit/iotit.go:61
github.com/xshellinc/iotit/vendor/gopkg.in/urfave/cli%2ev1.HandleAction
    /Users/daryl/Documents/goProjects/src/github.com/xshellinc/iotit/vendor/gopkg.in/urfave/cli.v1/app.go:485
github.com/xshellinc/iotit/vendor/gopkg.in/urfave/cli%2ev1.(*App).Run
    /Users/daryl/Documents/goProjects/src/github.com/xshellinc/iotit/vendor/gopkg.in/urfave/cli.v1/app.go:259
main.main
    /Users/daryl/Documents/goProjects/src/github.com/xshellinc/iotit/iotit.go:318
runtime.main
    /usr/local/go/src/runtime/proc.go:185
runtime.goexit
    /usr/local/go/src/runtime/asm_amd64.s:2197
```

tectiv3, 2017-09-03:
I can't extract http://dlcdnet.asus.com/pub/ASUS/mb/Linux/Tinker_Board_2GB/20170417-tinker-board-linaro-stretch-alip-v1.8.zip without CRC error. @makeupsomething can you?

makeupsomething, 2017-09-03:
@tectiv3 I had a problem extracting too. Using 7zip from the terminal worked for me.

tectiv3, 2017-09-03:
yeah, I found that post too, but 7z still reports crc error for me.

tectiv3, 2017-09-03:
@makeupsomething @yutashi Please test latest iotit with all supported boards.

makeupsomething, 2017-09-04:
Testing now

closed 2017-09-20 by tectiv3