xshellinc/iotit #101: Iotit windows 10 error

makeupsomething, 2017-09-27:
Maybe outdated but we got this error testing on windows 10

```
-------Showing last 10 lines-------
time="2017-09-08T11:24:14+09:00" level=info msg=DeviceInit args="[]"
time="2017-09-08T11:24:19+09:00" level=debug msg="SD flasher"
time="2017-09-08T11:24:19+09:00" level=debug msg=Prepare
time="2017-09-08T11:24:21+09:00" level=error msg="Error while running `which` : exec: \"which\": executable file not found in %PATH%"
time="2017-09-08T11:27:02+09:00" level=info msg=DeviceInit args="[]"
time="2017-09-08T11:27:09+09:00" level=debug msg="SD flasher"
time="2017-09-08T11:27:09+09:00" level=debug msg=Prepare
time="2017-09-08T11:27:11+09:00" level=error msg="Error while running `which` : exec: \"which\": executable file not found in %PATH%"
time="2017-09-08T11:28:37+09:00" level=info msg=DeviceInit args="[LOG]"
time="2017-09-08T11:28:37+09:00" level=fatal msg="LOG device is not supported"
```

tectiv3, 2017-09-27:
I need a win machine to replicate it.
@dbsparkle do we have a windows box that I could remote desktop to and test/fix that kind of stuff?

makeupsomething, 2017-09-27:
@tectiv3 The only windows machine we have is @SmevSmevson's

yutashi, 2017-09-27:
we have a laptop of windows10

makeupsomething, 2017-09-27:
@yutashi 
The laptop is 32 bit I think. So wont work.

tectiv3, 2017-09-27:
may be the windows is 32bit but the laptop isn't? Can I see it?

makeupsomething, 2017-10-31:
I think this issue is outdated now

closed 2017-10-31 by makeupsomething