xshellinc/iotit #102: System stops when I try to flash Raspberry

NanamiTakayama, 2017-10-02:
Here is my erroe log

```
Satone-no-MacBook:~ satoneyamanami$ ~/Downloads/iotit_0.3.3_macos_amd64/iotit -v
iotit version 0.3.3
Satone-no-MacBook:~ satoneyamanami$ ~/Downloads/iotit_0.3.3_macos_amd64/iotit
   [1] raspberry-pi
   [2] edison
   [3] nano-pi
   [4] beaglebone
   [5] colibri iMX6
   [6] esp
   [7] custom board
[?] Select device type: 1
[+] Flashing raspberry-pi
[-] Error:  raspberry-pi device is not supported
```

tectiv3, 2017-10-02:
O_o

yutashi, 2017-10-02:
@NanamiTakayama Can you please check it again with latest versin 0.3.4?

tectiv3, 2017-10-02:
@yutashi you are genius! @NanamiTakayama mapping.json format has changed. You are using and old version. Please do download the latest one.

NanamiTakayama, 2017-10-02:
@tectiv3 @yutashi 
I downloaded v0.3.4 with brew. and It works parfectlly.
sorry for interrupting you guys.

closed 2017-10-02 by makeupsomething

tectiv3, 2017-10-02:
Great!