xshellinc/iotit #103: Failed to flash colibri iMX6 from Windows

yutashi, 2017-10-06:
I tried to flash the board with Windows 10 and Powershell

```
PS C:\Users\xshell_admin\Downloads> .\iotit.exe
   [1] Raspberry Pi
   [2] Espressif ESP
   [3] Intel® Edison
   [4] Nano Pi
   [5] Beaglebone
   [6] Toradex Colibri iMX6
   [7] ASUS Tinker Board
   [8] Custom board
[?] Select device type: 6
[+] Flashing Toradex Colibri iMX6
[+] Using Angstrom LXDE Linux
[+] Added VirtualBox installation directory to system PATH
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x68 pc=0x7aedda]

goroutine 1 [running]:
github.com/xshellinc/iotit/device.(*colibri).exec(0xc0420ccba0, 0x8be28f, 0x44, 0xc0
        /home/ubuntu/.go_project/src/github.com/xshellinc/iotit/device/colibri.go:30
github.com/xshellinc/iotit/device.(*colibri).installTools(0xc0420ccba0, 0xc0420857b0
        /home/ubuntu/.go_project/src/github.com/xshellinc/iotit/device/colibri.go:10
github.com/xshellinc/iotit/device.(*colibri).Prepare(0xc0420ccba0, 0x80c980, 0xc0420
        /home/ubuntu/.go_project/src/github.com/xshellinc/iotit/device/colibri.go:35
github.com/xshellinc/iotit/device.(*colibri).Flash(0xc0420ccba0, 0xa8f9a0, 0xc0420cc
        /home/ubuntu/.go_project/src/github.com/xshellinc/iotit/device/colibri.go:90
main.main.func1(0xc0420606e0, 0xc0420606e0, 0xc042085887)
        /home/ubuntu/.go_project/src/github.com/xshellinc/iotit/iotit.go:61 +0x56
github.com/xshellinc/iotit/vendor/gopkg.in/urfave/cli%2ev1.HandleAction(0x819ea0, 0x
        /home/ubuntu/.go_project/src/github.com/xshellinc/iotit/vendor/gopkg.in/urfa
github.com/xshellinc/iotit/vendor/gopkg.in/urfave/cli%2ev1.(*App).Run(0xc042023860,
        /home/ubuntu/.go_project/src/github.com/xshellinc/iotit/vendor/gopkg.in/urfa
main.main()
        /home/ubuntu/.go_project/src/github.com/xshellinc/iotit/iotit.go:319 +0xf4f
```

the log is here:

```
PS C:\Users\xshell_admin\Downloads> .\iotit.exe log
Log location: C:\Users\XSHELL~1\AppData\Local\Temp\iotit.log
-------Showing last 10 lines-------
time="2017-10-06T10:29:39+09:00" level=info msg=DeviceInit args="[]"
time="2017-10-06T10:29:41+09:00" level=debug msg=Prepare
time="2017-10-06T10:29:47+09:00" level=error msg="Error while running `which` : exec: \"which\": executable file not fo
time="2017-10-06T10:29:47+09:00" level=debug msg=Prepare device=colibri
time="2017-10-06T10:29:47+09:00" level=debug msg="apk add dosfstools parted sudo e2fsprogs-extra coreutils libattr zip"
time="2017-10-06T10:30:21+09:00" level=info msg=DeviceInit args="[]"
time="2017-10-06T10:30:25+09:00" level=debug msg=Prepare
time="2017-10-06T10:30:31+09:00" level=error msg="Error while running `which` : exec: \"which\": executable file not fo
time="2017-10-06T10:30:31+09:00" level=debug msg=Prepare device=colibri
time="2017-10-06T10:30:31+09:00" level=debug msg="apk add dosfstools parted sudo e2fsprogs-extra coreutils libattr zip"
```

tectiv3, 2017-10-06:
did you install everything properly as stated in readme?

tectiv3, 2017-10-06:
https://github.com/xshellinc/iotit#installation
specifically git-for-windows and options from the example screenshot.

tectiv3, 2017-10-09:
@yutashi

yutashi, 2017-10-10:
sorry for being late. I will try it tonight

yutashi, 2017-10-10:
the above problem caused from no virtual box extension pack
it is solved now

yutashi, 2017-10-10:
and I got next error look like that.

```
PS C:\Users\xshell_admin\Downloads> .\iotit.exe
   [1] Raspberry Pi
   [2] Espressif ESP
   [3] Intel® Edison
   [4] Nano Pi
   [5] Beaglebone
   [6] Toradex Colibri iMX6
   [7] ASUS Tinker Board
   [8] Custom board
[?] Select device type: 6
[+] Flashing Toradex Colibri iMX6
[+] Using Angstrom LXDE Linux
[+] Checking virtual machine
[+] Starting virtual machine download
[+] Download iotit-box.zip   29.33 MiB / 29.33 MiB  100.00%
[+] Unzipping iotit-box.zip   136.00 MiB / 136.00 MiB  100.00% 0s
[+] Done
[+] Registering iotit-box
[+] Done
   [1] Use default vbox preset
   [2] Create a new vbox preset
[?] Please select an option: 1
[+] Selected virtual machine
        Name - iotit-box
        Description -
[+] Starting: /
[+] Starting download Angstrom LXDE Linux
[+] Download Colibri-iMX6_LXDE-Image_2.7b3+-20170630.tar.bz2 14.70 KiB / ?  150.56% 0s
[+] Uploading Colibri-iMX6_LXDE-Image_2.7b3+-20170630.tar.bz2 to virtual machine
[+] Extracting Colibri-iMX6_LXDE-Image_2.7b3+-20170630.tar.bz2
[-]  tar: invalid tar magic
[+] Configuring...
[+] Preparing: \
[+] Image ready
[+] Copying files: /
[-] Error:  exit status 1
```

```
-------Showing last 10 lines-------
time="2017-10-10T21:09:27+09:00" level=debug msg=Configure device=colibri
time="2017-10-10T21:09:27+09:00" level=debug msg="Creating tmp folder"
time="2017-10-10T21:09:27+09:00" level=debug msg="mkdir -p /tmp/isaax-sd/"
time="2017-10-10T21:09:28+09:00" level=debug msg="Running update.sh"
time="2017-10-10T21:09:28+09:00" level=debug msg="cd /tmp/ && ./update.sh -o /tmp/isaax-sd/"
time="2017-10-10T21:09:29+09:00" level=debug msg="cd /tmp/isaax-sd/ && zip -0 -r ../colibri_image.zip *"
time="2017-10-10T21:09:30+09:00" level=debug msg="ls -la /tmp/colibri_image.zip"
time="2017-10-10T21:09:31+09:00" level=debug msg="Downloading image from vbox" img=colibri_image.zip
time="2017-10-10T21:09:32+09:00" level=error msg="Sink: \\001scp: /tmp/colibri_image.zip: No such file or directory\n"
```

I've installed git-for-windows and the extension pack

tectiv3, 2017-10-10:
Something is wrong with image link, I think same error will be on any platform.

tectiv3, 2017-10-13:
yep, new image version. I've updated mapping in both this repo and the cloud.

yutashi, 2017-10-14:
I could flash on Windows 10

closed 2017-10-14 by tectiv3